armadsen/ORSSerialPort #38: sleep/wake handling doesn't actually work

razor-1, 2015-02-02:
ORSSerialPortManager attempts to register for notification of NSWorkspaceWillSleepNotification and NSWorkspaceDidWakeNotification in order to close ports and re-open them when the machine sleeps and wakes.  However, these are registered on the default notification center and thus are never fired.  

Per https://developer.apple.com/library/mac/qa/qa1340/_index.html they need to be registered on the NSWorkspace notification center.  Something like this:

```objc
    NSNotificationCenter *nw = [[NSWorkspace sharedWorkspace] notificationCenter];
	[nw addObserver:self selector:@selector(systemWillSleep:) name:NSWorkspaceWillSleepNotification object:NULL];
	[nw addObserver:self selector:@selector(systemDidWake:) name:NSWorkspaceDidWakeNotification object:NULL];
```

armadsen, 2015-02-02:
Nice catch. I'm quite sure this did work at some point... Anyway, I won't be able to get to this immediately. If you are able to submit a pull request, it would be most appreciated.

armadsen, 2015-02-03:
> Just realized I didn't changed dealloc to remove the observer. Not sure if that is important or not.

It’s not too important because ORSSerialPortManager generally is never deallocated. But I was just about to make that change :).

closed 2015-02-03 by armadsen