armadsen/ORSSerialPort #47: not compiling on OS X 10.7

azplanlos, 2015-03-25:
not compiling on OS X 10.7 due to NS_DESIGNATED_INITIALIZER in ORSSerialPort.h as discussed in pullrequest #46 comments.

azplanlos, 2015-03-25:
Also not compiling due to "-fobjc-arc is not supported on platforms using the legacy runtime"

armadsen, 2015-03-25:
The legacy runtime is only for 32-bit targets. ORSSerialPort has never offered support for 32-bit targets (well, not since before I published it as open source, anyway).

azplanlos, 2015-03-25:
but 32bit is activated as $(NATIVE_ARCH) includes 32bit on OS X 10.7. Never needed 32-bit targets anyway, but a change in the Framework Project file to 64bit Intel fixes this issue.

azplanlos, 2015-03-25:
also compilation of Framework Project is broken due to missing linking of IOKit

azplanlos, 2015-03-25:
and Cocoa.framwork

armadsen, 2015-03-25:
Oh, that makes sense. I'm not opposed to changing to explicitly setting the architecture to 64-bit in the framework project. Linking IOKit and Cocoa explicitly is fine with me too. These aren't necessary in modern versions of Xcode, but they shouldn't break anything either.

azplanlos, 2015-03-25:
yes, I think so eather. Shall I create a pull request for these changes?

armadsen, 2015-03-25:
Sure, that would be great.

armadsen, 2015-03-25:
Fixed by PR #48.

closed 2015-03-25 by armadsen

azplanlos, 2015-03-25:
thanks. Just found another one… [NSArray firstObject] does not exist on 10.7. As it was included in your latest commit it was not in my fork… Just realized after merging. Changed it to equivalent [NSArray objectAtIndex:0] and compiling fine again.