armadsen/ORSSerialPort #51: respondsToSelector: causes a bad access error

evanlooij, 2015-03-27:
ORSSerialPort cleanupAfterSystemRemoval uses respondsToSelector: to determine whether or not to send serialPortWasRemovedFromSystem: to the delegate. However, respondsToSelector: needs a reference to AppKit, which ORSSerialPort lacks. One solution is to import AppKit. On the other hand, the respondsToSelector: call is not really necessary since the delegate is specified as confirming to the ORSSerialPortDelegate protocol in which serialPortWasRemovedFromSystem: is a required method.
So, a simple check to see if the delegate has been should be sufficient, as in:
```javascript
- (void)cleanupAfterSystemRemoval
{
	if (_delegate)
	{
		dispatch_async(dispatch_get_main_queue(), ^{
			[self.delegate serialPortWasRemovedFromSystem:self];
			[self close];
		});
	}
	else
	{
		[self close];
	}
}
```

armadsen, 2015-03-27:
Hi Elise, thanks for contributing.

`-respondsToSelector:` is not part of AppKit. It's an NSObject method, and certainly doesn't require AppKit to be linked. You can verify that by compiling and running [this tiny test program](https://gist.github.com/armadsen/cf98ceb3f4d39c5347c1).

For what it's worth, while in this case `-serialPortWasRemovedFromSystem:` is required, all the other `ORSSerialPortDelegate` methods are optional, so I do have to check for those using `-respondsToSelector:`.

Is it possible for you to send an example program that's crashing? Without seeing your code, my best guess is that the `ORSSerialPort` object's delegate is being deallocated before `-cleanupAfterSystemRemoval`'s dispatched block is executed, causing a crash. ORSSerialPort uses `unsafe_unretained` for its delegate property in apps whose deployment target is earlier than OS X 10.8. The solution is to either set the deployment target to 10.8 or higher, if you're able, or to explicitly set the port's delegate property to nil before the delegate is released.

evanlooij, 2015-03-27:
[image]

evanlooij, 2015-03-27:
[image]

evanlooij, 2015-03-27:
I've isolated the test that was crashing into a separate project, but I'm not sure how to get it to you.

armadsen, 2015-03-27:
Hi Elise. You can email me at andrew@openreelsoftware.com.

armadsen, 2015-03-27:
That documentation page is interesting. I'm not sure why it lists NSObject in AppKit. I don't see that in my copy of Xcode, and the [online documentation](https://developer.apple.com/library/mac/documentation/Cocoa/Reference/Foundation/Classes/NSObject_Class/index.html#//apple_ref/doc/uid/TP40003706) says it's in Foundation.

In fact, NSObject, including its implementation of `-respondsToSelector:` is part of the Objective-C runtime itself. You can even view the source: http://opensource.apple.com/source/objc4/objc4-532.2/runtime/NSObject.mm .

The test program in my last comment will also let you verify that AppKit is not required to use `-respondsToSelector:`:

[image]

Note that otool only reports that the program is linked with Foundation, CoreFoundation, libSystem, and libobjc.

armadsen, 2015-03-27:
Presumably fixed by updating to latest ORSSerialPort revision to take advantage of weak delegate on ORSSerialPort.

closed 2015-03-27 by armadsen