armadsen/ORSSerialPort #59: Serial  Data in a Swift Playground

samuraipapa, 2015-07-05:
This code is great. I've been using it in a Swift OS X project, with Arduino.  

I'm trying to graph an incoming sensor value from Arduino's Analog Input Pin.

I works in my OS X  app (with your code), very well.  Is there an easy way to port ORSSerialPort.h functionality into a Swift Playground?  A hello-world playground would be very appreciated. 
I'd be happy to fill-out the rest of the code, to graph the Analog data in a playground, in realtime.

armadsen, 2015-07-05:
I'm glad you're finding ORSSerialPort useful. Unfortunately, I don't think it's currently possible to use ORSSerialPort in a Swift playground because OS X Swift playgrounds are sandboxed, and the set of entitlements they get is not configurable. Network access is permitted, but other prohibited-by-default actions, including arbitrary file and serial port access are not. Sandboxed applications must declare the com.apple.security.device.serial entitlement in order to use ORSSerialPort.

I have filed [Radar #21677248](http://www.openradar.me/21677248) with Apple regarding this limitation. Feel free to file a duplicate with them if you'd like to add your voice.

You can download an example playground that should work if not for this limitation [here](https://www.dropbox.com/s/w49prikhuzkgjmc/SerialPlayground.zip?dl=0).

samuraipapa, 2015-07-05:
Thanks for the info.  Great to know. 
Yes, I was able to replicate your error with your Playground file.  It looks like a great Playground btw.  When it works with ORSSerial, will be even better.  ; )

The Radar is a good idea.

armadsen, 2015-07-06:
Since there's nothing I can do on this until/unless Apple addresses the underlying problem, I'm going to go ahead and close this. 

If Apple makes it possible, I will definitely add a playground to the ORSSerialPort examples.

closed 2015-07-06 by armadsen

samuraipapa, 2015-07-06:
Perfect. Thanks for the help anyway.