armadsen/ORSSerialPort #64: Kernel panic when closing an FTDI port using the Apple driver, when another app has an open port

armadsen, 2015-08-01:
User report: 

>Seems to occur only when RUMlogNG is also running and has other serial ports open, but not the one I have open with ORSSerialPortDemo. Happens on the close() call. Doesn’t happen every time, maybe once in three or four calls to close().
>
>The call to close() hangs, and the entire Mac crashes after a few seconds, This is on OSX 10.9.5 and I compiled the demo program unchanged with Xcode 6.2 using the 10.10 SDK.
>
>When OSX does not crash, the close() call happens quickly, as you would want it to. But when RUMlogNG 1.2.5 is running with (I assume) the latest version of ORSSerialPort, it will crash OSX fairly often.

cbjohnson46, 2015-08-08:
Further testing with 2.0 shows that the kernel panic will occur under 10.9.5 without any other app having opened a serial port. In my testing, the panic occurs once in three close() calls.

armadsen, 2015-08-09:
Confirming that I can reproduce this. I have a hunch about what's causing it, and am working on (hopefully) a fix now.

armadsen, 2015-08-09:
I've committed a fix for this. According to the documentation for `dispatch_source_cancel()`:

>[T]he cancellation handler is a convenient place to close or deallocate [file descriptors or mach ports]. Note that it is invalid to close a file descriptor or deallocate a mach port currently being tracked by a dispatch source object before the cancellation handler is invoked.

I'm a bit surprised that "invalid" means kernel panic! Nevertheless, I was doing this wrong, and changing to closing the underlying file descriptor in the read dispatch source's cancellation handler seems to have solved the problem.

After this change, in my testing, opening and closing an FTDI port (using FTDI's driver) several dozen times has not provoked a kernel panic. The port is also closed very quickly. I'd appreciate your testing this @cbjohnson46. I'll wait to close the issue as fixed until I hear back from you. Thanks!

armadsen, 2015-08-09:
Do note that this change has one side effect for users of ORSSerialPort. `-close` is no longer synchronous. It return immediately, and the actual port closure happens later. The port closure is still reported by a call to the ORSSerialPort delegate method `-serialPortWasClosed:`. `-close` has never been documented to be synchronous, and a delegate method to report closure has always existed in order to facilitate changing to an asynchronous model, but for all previous versions of ORSSerialPort, `-close` has in fact been synchronous. This change should be noted in the release notes for version 2.0.

cbjohnson46, 2015-08-09:
Can no longer cause a kernel panic with the most recent commit to 2.0. The call to close() happens quickly, but further testing with multiple ports open by the same app or even different apps seems appropriate. I have only tested the most recent change with the "demo" app driving a single serial device.

armadsen, 2015-08-10:
I'm closing this as fixed now. @cbjohnson46, if in your further testing, you run into trouble, please let me know, and I'll reopen it.

closed 2015-08-10 by armadsen