armadsen/ORSSerialPort #67: Device PID/VID

knseir, 2015-08-13:
Hi
how can I get a ORSSerialPort Object's PID or VID ? 
I've looked through the Wiki, couldn't find anything 
Thank you

armadsen, 2015-08-13:
Hi knseir,

I've added a page to the wiki about this: https://github.com/armadsen/ORSSerialPort/wiki/Getting-Vendor-ID-and-Product-ID

It's based on this Gist, which I've had around for some time: https://gist.github.com/armadsen/5131559.

Let me know if you have any questions.

closed 2015-08-13 by armadsen

knseir, 2015-08-14:
thank you ! That worked

cmaglie, 2015-12-03:
@armadsen 
would it be possible to include this snippet in the library? is there a reason to not include these properties upstream and make them available directly from the library?

armadsen, 2015-12-03:
@cmaglie thanks for the comment. My reasons for not including it in the API itself are:

- Serial ports are not _necessarily_ USB devices (though at this point, they probably are in 99.9% of cases)
- These properties are not really fundamental to 'serial port programming'.

That said, I would certainly consider including them if they're useful for enough people. What's your use case?

cmaglie, 2015-12-04:
Hi @armadsen, thanks for the quick feedback!

The typical use case is auto-detection of the attached device.

Almost all devices that implements a virtual USB serial port (modems, BTLE dongles, dev boards etc.) have their unique VID/PID that can be used to:

1. detect when the device has been connected.
2. automatically select the correct serial port and start communicating with the device with the right protocol

A big advantage is that this kind of detection is performed **without active probing** just using the information exchanged out-of-band by the USB driver.

My particular use case is for the Arduino IDE to show the connected Arduino board near each available serial port:

[image]

This is a very popular and really appreciated feature.

We are currently doing it by polling Mac OS X's `system_profiler` to gather USB information, using this library we could make with a native call from Java, gaining some clock cycle.