armadsen/ORSSerialPort #71: receiveBuffer can grow unbounded

dolfs, 2015-09-05:
When using ORSSerialPort with a device that sends data repeatedly, without stopping and not using request/response the receive buffer keeps growing without bounds.

One may "consume" data in the didReceiveData delegate method (which is only "fed" the newly received data), the internals then append the data to the receive buffer. The buffer is only ever cleared/emptied when issuing a command. I suppose issuing an empty command with a nil completion block might,effectively clear that buffer, but it seems rather circumspect.

Since the buffer is also never directly readable I think the appending should only happen if there is a pending request.

dolfs, 2015-09-05:
I think the answer is actually more complicated. There is an inherent producer consumer problem here, with the producer being the thread attempting to read from the serial port and the consumer is the code trying to read from the buffer.

The producer and consumer should share a bounded buffer of a certain size and once full the producer should be blocked from adding to it until the other size at least consumes some. This means the producer should never attempt to read more characters from the file descriptor than it can fit in the buffer, or it can read as much as it wants, but trying to append to the buffer should suspend that thread.

On the receiving end we have two different models of reading data. The "plain" method should be reading from the other end of the buffer.We should inform the delegate if the amount of available data in the buffer has changed and provide a method of "peeking" at it. There should be a separate method to let a user of the ORSSerialPort consume bytes from the buffer and it should work by requesting a certain amount, but returning less if data is not available. This will, eventually, empty the buffer, allowing the producer to restart.

Using this approach, no characters will ever be lost and we can either consume the data piece by piece if we wish, or use a request/response type approach. If the request/response implementation buffers all the data consumed while looking for completion (separate from the receive buffer), the size of the receive buffer will not fundamentally limit what can be received in a single response, but it should be sized to absorb the typical "harmonica" effect between producer and consumer. The latter is application dependent, so it should be possible to configure the size of the bounded buffer.

armadsen, 2015-09-05:
@dolf, thanks for the write up. I'll have to digest what you're suggesting a little before providing a constructive response.

As for the fundamental problem of the receive buffer being able to grow without bound, that needs to be fixed one way or the other. I believe this is currently only a problem with the request/response API (and the new packet parsing API in 2.0). ORSSerialPort's original paradigm (pre-1.5) was simply to deliver data as it is received, via the `-serialPort:didReceiveData:` delegate method, leaving any buffering of incoming data up to the client code. This mechanism is still available, of course.

armadsen, 2015-09-06:
This should be fixed by #77.

closed 2015-09-06 by armadsen