armadsen/ORSSerialPort #72: Suggest adding block based alternative to delegate approach

dolfs, 2015-09-05:
For a lot of applications it will be a lot easier to handle responses to requests, for example, but accepting a completion handler (which would receive what the delegate receives). This avoids having to relate what is received in the delegate method to an original request (it can be done now, but requires the userInfo object).

For error handling it is hard to do correctly as when didEncounterError is called there is no "context". If this were a callback, the callback itself would provide the context.

It would, likewise, be very useful for open (and close) to have a completion handler block option as well.

armadsen, 2015-09-05:
I see some value in this, and in fact considered it when first starting on ORSSerialPort. However, I have mixed feelings about it. In any case, major API changes will have to wait until ORSSerialPort 3.0.

However, I think you could fairly easily wrap (or possibly subclass) ORSSerialPort to "convert" its delegate callback system into a block based approach, at least for `-serialPort:didReceiveResponse:toRequest:`.

armadsen, 2015-09-07:
I'm not going to do this for now, but rethinking things in this area in light of changes since 2011 when ORSSerialPort was original developed may make sense for 3.0.

closed 2015-09-07 by armadsen