armadsen/ORSSerialPort #73: Suggest adding a method to just read input until "complete"

dolfs, 2015-09-05:
This would effectively be the same as sendRequest with an empty command, but a completionDetectionBlock.

While the sendRequest with empty command might work, semantically it is resulting in confusing code.

armadsen, 2015-09-05:
Hi @dolfs, as mentioned in my comment on #74, the 2.0 branch provides a new packet parsing API.

Using that, I think you can do what you're talking about in a more elegant way. With that API, you no longer have to provide an empty request to packetize incoming data. It is designed so you can tell the port to listen for packets matching a descriptor. The port will continue to process and deliver packets indefinitely, allowing for receipt of multiple packets, but you can also stop listening at any time. So if you only want to receive a single packet, you can start listening for a certain packet, then as soon as it is received, tell the port to stop listening.

armadsen, 2015-09-07:
I consider this fixed by #55. (But see #74 for more thorough discussion.)

closed 2015-09-07 by armadsen