armadsen/ORSSerialPort #74: Request API can return too much data

dolfs, 2015-09-05:
I have a device that outputs serial data representing some sensors quite fast. Issuing a request with an empty command just so I can read one line sometimes grabs multiple lines.

The source of the problem is that when the ORSSerialRequestResponseEvaluator gets called it can only return yes or no, but not how many bytes of the data (in the receive buffer) were needed to produce a yes. Consequently when a yes is produced, the whole receive buffer is discarded. The problem is most easily seen in this example because every line ends with a return so it is easy to eventually deliver a yes, but is not unique to this situation.

My check basically looks for a carriage return ending the data. The data, however, arrives quick enough that sometimes on the first callback the buffer has captured more than one line, ending in a partial line. Since I check for the data to end with a "\r" its is not seen and we capture some more until, at some point, we find a return at the end of some line. All captured lines are returned.

It is not hard to check for the first occurrence of the return instead, but even that does not quite work because if on first check a line and as half was captured, the whole line and a half gets delivered. I only need up to the return so would discard the rest, which leaves the partial line lost forever.

Of course this can be coded around, but it would defeat the purpose of the existence of the request/response API.

I suggest an alternate response evaluator that can return how many bytes were needed to produce a YES. The response can then be constructed from those bytes and the rest left in the buffer. This also means, however, that the sendRequest implementation should not clear the buffer.

dolfs, 2015-09-05:
I have implemented the suggested approach by letting the completionDetector return an NSRange instead of a boolean. I am (ab)using the userInfo as a mutable and shared dictionary between the request and the didReceiveResponse delegate (this will be thread safe) and store than range (when the detector returns true) so that it can be used in the delegate method to extract only the matched data (do not have to call the extraction method a second time like is done in the request/response sample code).

I am not quite setup to produce a pull request for this, but I am happy to share code via private communications if you want it earlier.

armadsen, 2015-09-05:
Hi @dolfs. Thanks for the feedback. This is indeed a problem. However, I'd encourage you to checkout the [2.0 branch](https://github.com/armadsen/ORSSerialPort/tree/2.0) of the project. One of the new features in 2.0 is a packet parsing API (see #55) that can be used to receive incoming packets that are _not_ responses to an outgoing request.

The packet parsing code fixes this problem, though not in quite the same way you propose. Rather, it uses a reverse-search parser that can handle multiple packets coming in at once. I have also just added a unit test for exactly this scenario.

The request/response API has been reimplemented on top of packet parsing, so it benefits from this fix too.

Do note that as documented, an `ORSSerialPacketDescriptor` with a nil prefix, and only a suffix will be considered to describe packets that consist solely of the suffix. So in your case, you'd just get the carriage return. You should instead implement a response evaluator block that validates not only that the data ends with a carriage return, but that it matches the other criteria specific to the packets you're receiving (e.g. length, checksum, etc.).

dolfs, 2015-09-07:
I have taken a look, but unfortunately, this will not work for me either. There are several problems with the approach, one of which I saw you just fixed.
First is that each detector keeps its own buffer,which can grow quite large if not optimized. Second, detectors cannot work incrementally, which makes it expensive.

The biggest problem, however, is fundamentally the same as with the 1.0 branch. You start up a polling type process that actively takes from the serial port (as long as data is available) and "pumps" it in the rest of your machinery.

In an application such as mine, where the other side of the serial port is sending data continuously this approach (which by designs never blocks in this scenario) consumes unbounded system resources (CPU and buffer). I experienced this the other day when I decided to see if I could get around my issues by ignoring the request/response and simply receive data in "didReceiveData". The result was the the polling process was queuing up async calls to the didReceiveData delegate at tremendous speed in its loop. Since the loop essentially never quit (because data is always coming), the async tasks on the main queue never got run (at least not while I was waiting).

Even if they would run, they would dominate the main queue and all UI stuff would come to a halt.

Besides the fact that for applications such as these I think callbacks should be used rather than delegates (so you have context when you get called in the form of captured variables), the callbacks (blocks or delegates) should be called on the background thread and be responsible for switching to the main thread themselves. Often it is not needed forcing an unnecessary trampoline where the main thread versions would immediately queue something up on a concurrent queue.

Based on this, I have decided that your approach, while possibly/likely fine for anyone who has a more well behaved device that does not send unbounded data and trigger the issue, is not going to work for me. It also does not sound that would change anytime soon. So I decided to roll my own.

In my approach, the polling loop still exists, but it dumps data into a bounded buffer, meaning it has a fixed size in bytes, and when the buffer is full, the polling task blocks when trying to add more. Data is removed from the buffer by one or more other threads which will block if no data is available, and which will unblock the appending poller when at least one byte is taken out (because the buffer is no longer full). I have that working just fine.

While I have not (yet) implemented this, this can also be used with flow control so that the other end stops sending until we're ready for more.

It is now easy to build a pipeline where the buffer, for example, is being read by a process that divides everything in lines and only hands complete lines to whoever is next, simplifying processing down the line. That's just a simple example, filtering, matching, etc. is also possible. Ultimately I will hook up reactive signals to this end.

Anyway, I want to thank you for your efforts on this project because, at least initially, it allowed me to do some simple serial processing real fast and I am sure it will be fine for many others. In due time I will see if I can open source my approach. Time pressures don't allow me to do that right now. I'd be happy to discuss more if you have questions, but I think we should do that over email.

armadsen, 2015-09-07:
Thanks for the detailed reply, @dolfs. Everything you said makes sense. ORSSerialPort started out as my modern replacement for AMSerialPort in a fairly simple amateur radio control app. Some of the early design decisions reflect this beginning, including dispatching data in small chunks to the main queue where it will be used to pretty directly update UI. Of course, this does have implications for the kind of scenario you're dealing with.

I would be interested in finding out more about the solution you've implemented, if for no other reason than that I'd like to learn from your approach. You've also got me thinking of possible solutions to this problem in a future version of ORSSerialPort (where of course API backwards compatibility is a concern).

I don't have your email address, but you can contact me at andrew@openreelsoftware.com.

For now, I'm going to close this and the other issues you opened. Thanks again.

closed 2015-09-07 by armadsen