armadsen/ORSSerialPort #85: Add support for changing number of data bits

armadsen, 2015-09-28:
Currently, ORSSerialPort configures ports for 8 data bits with no way to change this. Making this configurable would be nice, and is probably long overdue (though somewhat surprisingly, I've never had anyone ask for this...).

csann, 2015-09-28:
I've worked with industrial control systems for more than 30  years...during the era when serial ports were used a lot. I can't ever recall setting the number of data bits to anything but 8.

armadsen, 2015-12-12:
Postponing this at least until after 3.0 for now.

apelican, 2016-10-19:
I very definitely ran into this interfacing with a piece of hardware that defines its connection as 7 data bits, MARK parity, and two stop bits.