armadsen/ORSSerialPort #89: Add support for parsing packets consisting of arbitrary data followed only by a terminator.

armadsen, 2015-10-07:
ORSSerialPacketDescriptor's design was driven by a desire to design a system that is robust against dropped or corrupted data. As such, it currently only supports describing and parsing full-specified packets. That is packets whose start and end (at least) can be clearly determined from the data alone.

However, it is fairly common for serial data to come in line by line where the data is arbitrary bytes followed by a terminator. A complete packet in this case is just all the data that has come in since the last packet finished (ie. terminator was received). As it stands, ORSSerialPacketDescriptor and the packet parsing API can't handle this kind of data.

It may require adding a special case for this in the packet parsing logic.

csann, 2015-10-07:
You might also consider allowing a packet to be described by a suffix and/or prefix and a fixed number of bytes. For example, 22 bytes starting with a prefix of <12 d3 6a>. This can be done with a PacketEvaluator however that requires a lot of code for such a simple requirement.

armadsen, 2015-10-07:
Thanks for the comment, Clark. I agree. However I do want any solution for this to be more general. This issue was partly prompted by a user of ORSSerialPort writing an app that receives packets where the length is arbitrary.

nathanntg, 2016-02-27:
I'd definitely put in a plug for something along these lines. I've been interacting with an Arduino sketch that has a predetermined protocol, and it uses just a packet terminator (\r\n), so I have not been able to make it work with `ORSSerialPacketDescriptor`. I am going to try and implement a work around myself, which I will mention in a pull request, although it may not be the best solution. And I would totally understand if you opt not to merge it given your 3.0 rewrite.

(Thanks for the great framework!)