armadsen/ORSSerialPort #90: PacketDescriptor using Regex

pjmnoble, 2015-10-15:
Hi, thought I submitted this yesterday but see no sign - my apologies if I'm repeating the message instead of just waiting for a moderator to approve it.

I have the NSRegular expression below used in a packetDescriptor to validate  a coordinates string such as "167628,79009\r" delivered via a serial port using ORSSerialPort requestResponse API. The expression, however leads to the delegate serialPort: didReceiveResponse: data containing "8,79009". 

If I replace the regex with "(\\d{5}),(\\d+)\r" it works and I get "167628,79009" (so ORSSerialPort is building the correct string) but this will only be useful when the coordinates delivered are a 5 digit number. If I use d{1,5} I get the same result as when using the line-start anchor.

The regex is ignoring the anchors. Any Suggestions ?

Code for the regex and packetDescriptor used

    coordinatePacketRegex = [[NSRegularExpression alloc] initWithPattern:@"^(\\d+),(\\d+)\r"
                                                                   options:NSRegularExpressionAnchorsMatchLines
                                                                       error:&regexError];

    coordinateDescriptor = [[ORSSerialPacketDescriptor alloc] initWithRegularExpression: coordinatePacketRegex
                                                                    maximumPacketLength:12
                                                                      userInfo:Null;

armadsen, 2015-10-15:
Hi pjmnoble,

Thanks for opening this issue. Fundamentally, this is the same issue as #89. Packets are parsed one byte at a time looking backwards from the most recently received byte. The smallest chunk of data that matches the regex is considered to be a valid packet. This is to prevent bad data sitting at the beginning of the buffer from being interpreted as being part of the packet.

I'm still considering the best way to solve #89. I'm glad you found this, because it's a case I hadn't thought of. It seems like it might make sense for regex-based packet descriptors to be matched against the entire buffer. If not always, at least when the regex contains a line-start anchor.

In your case, is it true that the coordinates are always 5-digit numbers (your example shows the first value as a 6-digit number...)? If so, I think the solution you found is a fine workaround. If not, it helps me prioritize a fix for this :)

pjmnoble, 2015-10-16:
Thanks for the reply. the (\d{5}),(\d+)\r option does not work for me because coordinates will often be supplied as a shorter string (when the numbers are smaller - eg 317,803 where the first coordinate won’t match)

I’ll use the data received selector to build the string and implement my own buffer - I have already got this logging how the data arrives and it builds the whole string perfectly. In the above instance I get:
3
31
317
317,
317,8
317,80
317,803
at which point the didReceiveResponse method is fired returning “7,803” which surprises me because when I feed the string 317,803 (the string I assume your system is building too) to the regex, it matches 317 and 803 not 7 and 803

Anyhow I’ll reset the buffer when I get a \r and see how that works.

Despite this, ORSSerial really makes life much easier - thanks

PJ
pjmnoble@mac.com

S4mw1s3, 2016-02-20:
I just spent quite some time figuring out why my regex wasn't capturing all the bytes I wanted. Then I found this issue here. Painful... Guess I'll also need to implement my own buffer for now.
Great lib though! :)

armadsen, 2016-12-12:
Closing as a duplicate of #89, where I'll track work on a fix for this.

closed 2016-12-12 by armadsen