armadsen/ORSSerialPort #91: Issue with the /dev/tty -> /dev/cu translation

raphaelschaad, 2015-11-28:
In `-[ORSSerialPort open]` I get a -1 descriptor when it tries to use the translated `self.path`. If I pass the original path (an FTDI cable) to `int open(const char *, int, ...)` I get a valid descriptor. Why do I have to hack the code like this?

```objc
…
int descriptor=0;
descriptor = open([/*self.path*/@"/dev/tty.usbserial-FTH7PIOR" cStringUsingEncoding:NSASCIIStringEncoding], O_RDWR | O_NOCTTY | O_EXLOCK | O_NONBLOCK);
if (descriptor < 1)
{
…
```

Thanks!

armadsen, 2015-11-28:
Hi Raphael,

I think this is probably related to #41. I don't quite understand why opening the cu path wouldn't work when opening the tty path doesn't. I've never seen that myself. Can you give me any more details on the hardware you're using?

raphaelschaad, 2015-11-28:
Do you mean "when opening the tty path *does*"? (because it does)

I'm using an FTDI cable mapping USB to serial connected to a custom board with a programmed AVR chip (ATTiny44a). Everything works if I apply the posted hack, but it's of course weird/ugly.

The cu-variant does show up on my system:

```
$ ls /dev/tty.usb*
/dev/tty.usbserial-FTH7PIOR
$ ls /dev/cu.usb*
/dev/cu.usbserial-FTH7PIOR       
```

armadsen, 2015-11-28:
Yes, sorry for the typo. What happens if you change line 160 of ORSSerialPort.m to the following?

`NSString *bsdPath = [[self class] bsdDialinPathFromDevice:device];`

Also, have you implemented the `ORSSerialPortDelegate` method `-serialPort:didEncounterError:`? Does that get called with an error when trying to open the port, and if so, what is the error code?

raphaelschaad, 2015-11-29:
Changing L:160 will keep self.path at "/dev/tty.usbserial-FTH7PIOR" (vs. the cu-version).

I also implemented all delegates and reverted all changes to ORSSerialPort to monitor what's going on and an interesting thing is happening: I currently don't get the error anymore (i.e. it works with the cu-version too) w/o changing anything on the setup (system, cables, devices, embedded).

raphaelschaad, 2015-11-30:
Yeah, it "just works" repeatedly now and I don't see anything that would have changed.

armadsen, 2015-11-30:
That's frustrating, @raphaelschaad. Without any action to take on this, I'm going to close it for now. If this crops up again and you're able to provide any illumination, let me know. Otherwise, I'm just going to figure the only thing to do for now is to implement #41, which I still plan to do.

closed 2015-11-30 by armadsen

raphaelschaad, 2015-11-30:
That sounds totally reasonable. Thanks for the fast responses. I will report back if a problem occurs again.