armadsen/ORSSerialPort #92: Rewrite ORSSerialPort itself in Swift

armadsen, 2015-12-07:
This is a major change for 3.0. My intention is to maintain Objective-C compatibility and keep breaking API changes to a minimum.

armadsen, 2015-12-12:
Don't close this without testing sleep/wake and close-on-quit functionality in the new Swift `SerialPortManager`.

kouky, 2015-12-21:
Wow, this would be amazing.

armadsen, 2015-12-21:
:+1: 

I'm well underway. The code in the 3.0 branch includes all classes except `ORSSerialPort` itself rewritten in Swift. I'm in the process of rewriting `ORSSerialPort` (aka. `SerialPort`) itself now.

Early feedback would be most welcome.

kouky, 2015-12-21:
@armadsen yes I can see you've been busy, wish I saw this earlier so I could have helped in the conversion.

marciok, 2016-03-07:
+1 
Looking forward, for the rewrite.

ladislas, 2016-09-25:
how can we help?