armadsen/ORSSerialPort #95: Documentation error?

kirk-kerekes, 2016-01-15:
In-code comments for `ORSSerialPacketDescriptor` assert:

```objc
*  Note that this API is intended to be used with data that is sent by a serial device periodically,
*  or in response to real world events, rather than in response to serial requests sent by the computer.
*  For request/response protocols, see ORSSerialRequest, etc.
```

-- but the `ORSSerialRequest` API *uses* `ORSerialPacketDescriptor`. 

I assume that this is obsolete documentation, and that the above Note should be deleted.

armadsen, 2016-01-15:
Hi @kirk-kerekes,

Thanks for the feedback. This documentation is referring to API usage by an end-user of ORSSerialPort. It is true that `ORSSerialRequest` uses `ORSSerialPacketDescriptor`, but that's an internal implementation detail.

My point is that a client user of ORSSerialPort will benefit from using the request/response API rather than the (lower-level) packet descriptor API when their use case (structured request/response protocol) matches up with its intended use.

Perhaps the note could be reworded to be more clear?

kirk-kerekes, 2016-01-15:
Yes, something like " if you are doing request/reply transactions, use the ORSSerialRequest API, which uses the ORSSerialPacketDescriptor as part of its API"

Kirk Kerekes 
(iPhone)

armadsen, 2016-01-15:
Ahh, right. For some reason, I was overlooking the fact that ORSSerialRequest takes ORSSerialPacketDescriptors as part of its public API (that wasn't always the case). I'll make a change to improve this documentation.

Thanks again.

closed 2016-01-16 by armadsen