armadsen/ORSSerialPort #97: CPU utilization outside of Xcode

davecrumbacher, 2016-02-11:
I have an app that uses ORSSerialPort and it works just fine when running it within Xcode.  However, if I try to run app on its own, my CPU is pegged and the app becomes unresponsive.  Any suggestions?

armadsen, 2016-02-11:
Hi Dave,

This isn't something I've heard reported before, nor do I know off the top of my head what would cause it.

Does it occur only when the app is built in Release mode, or do you see it when running a Debug build of the app outside Xcode? Are you able to run the app, then attach the debugger to see what's using CPU cycles? What about when you run in instruments?

Let me know. I'm also happy to take a look at the project itself if you're able to share it. You can email me at andrew@openreelsoftware.com

davecrumbacher, 2016-02-11:
Hi Andrew.  It happens with both Release and Debug builds outside of Xcode.  I have not done any debugging, but will let you know if I find anything.  What do you mean by running "in instruments"?  Thanks.

armadsen, 2016-02-11:
I meant running your app in [Instruments](https://developer.apple.com/library/mac/documentation/DeveloperTools/Conceptual/InstrumentsUserGuide/) which comes with Xcode. If you're lucky, it would hopefully allow you to inspect exactly what's using up so much CPU time without masking the bug the way running directly in Xcode is doing. Just a hunch that it could be helpful.

davecrumbacher, 2016-02-12:
Okay, I'm not familiar with Instruments, so I will take a look and see if I can get some useful information.  Thanks.

csann, 2016-02-12:
Checkout this intro to Instruments.  

http://www.raywenderlich.com/97886/instruments-tutorial-with-swift-getting-started

armadsen, 2016-06-12:
Closing this for no activity. If you're still seeing this or have further questions, let me know.

closed 2016-06-12 by armadsen