armadsen/ORSSerialPort #98: ORSSerialPortManager's kIOPublishNotification not firing in sandboxed app

evanlooij, 2016-02-13:
I'm working with an Arduino board connected via USB. When I run the example project ORSSerialPortDemo, the list of available ports is automatically updated when I plug in the board (while the app is already running). However, when I use the same technique in my own sandboxed app (an NSArrayController bound to ORSSerialPortManager.availablePorts) the list is never updated. 
I put an NSLog in -ORSSerialPortManager's serialPortsWerePublished: and in the sandboxed app it never gets called when I plug in the board while the app is running, in ORSSerialPortDemo it shows up in the log.
My conclusion is that IOKit's kIOPublishNotification does not fire in a sandboxed app. I'm compiling and running ORSSerialPort as a framework. 
The framework has its own entitlements: 
- com.apple.security.app-sandbox
- com.apple.security.inherit

The app has the following entitlements:
- com.apple.security.app-sandbox
- com.apple.security.device.serial
- com.apple.security.device.usb

The app builds the ORSSerialPort framework as a dependent target, then embeds it. Any ideas on how to deal with this?

evanlooij, 2016-02-13:
The console shows no sandboxd messages when the sandboxed app is run, unlike ORSSerialPortDemo

armadsen, 2016-02-13:
Hi Elise,

This isn't something I've seen before. I've used ORSSerialPort in a couple sandboxed apps, including the latest version of Aether, the app it was originally developed for, and haven't had this issue. For fun, just now, I turned on sandboxing for ORSSerialPortDemo and confirmed that the ports list correctly updated when plugging in a USB to serial adapter.

A few questions to get started troubleshooting:

- Does the behavior change if you plug in a USB to serial device? In other words, is it specific to your Arduino board?

- What kind of Arduino is it? (I have a few around here, and could test with them if I have the same type and it makes sense to do so.)

- Does the problem persist if you turn sandboxing off for your app?

evanlooij, 2016-02-17:
Thanks for the troubleshooting tips, I just now got it working. I first verified that sandboxing worked with your ORSSerialPortDemo. When it did, I went back, turned off the sandboxing and cleaned and rebuilt first the dependent targets and then the main target. When the notifications came through, I turned on sandboxing and now it is working. I had done the cleaning & rebuilding last week (more times than I care to remember), but that was before I realized that each target should have its own entitlements file with the com.apple.security.inherit setting. 
Also, yesterday I did some spring-cleaning in my developer certificates and keychain app. Anyway, somehow I stumbled upon the solution.

closed 2016-02-17 by evanlooij