armadsen/ORSSerialPort #99: Doesn't seem to work on latest veresion of OSX and Xcode

haiduc32, 2016-02-20:
I'm trying to run the CommandLineDemo (Swift version) connected to an Arduino (Leonardo R3) that streams data continuously over serial. 

Nothing gets read. I open the serial monitor in Arduino IDE and the data is being received continuously.
If I start the CommandLineDemo, and then start the Arduino IDE serial monitor, the XCode breakpoints get heat and it starts to read the data from the Arduino (although corrupted since some is being read by the Arduino serial monitor and some by the Swift application).

I've also tried an older project of mine that was working, and same problem. Tried to create a bear minimum project, same problem.

Ive tested it on:
OSX 10.11.3
Xcode 7.2.1

Tested also on a second Macbook with the same XCode, but not sure about the OSX version.

any ideas?

armadsen, 2016-02-23:
Hi @haiduc32,

Try setting the RTS line high as I've done in the RequestResponseDemo and PacketParsingDemo example projects:

```
- (void)serialPortWasOpened:(ORSSerialPort *)serialPort
{
	serialPort.RTS = YES; // Add this
	printf("Serial port %s was opened", [serialPort.name UTF8String]);
	printPrompt();
}
```

I never investigated deeply why this was required, but it definitely was required to make receiving data from my Arduino Esploras work on 10.11.

haiduc32, 2016-02-25:
Thanks for that. I'll be able to test this fix only next week, and will let you know how it works out.

haiduc32, 2016-03-10:
So I got around to testing another board, a SparkFun RedBoard (UNO bootloader) and it worked without any changes. I guess the Leonardo does something different. Now I don't have the Leonardo to test the suggested fix, but I hope I get it back tomorrow and will report on it too.

haiduc32, 2016-03-11:
I got the Leonardo board back today. I tested it without the RTS and it doesn't work. Adding the RTS = true does the trick.

Thanks for the help, and hope that helps anyone else that stumbles on that problem.

closed 2016-03-11 by haiduc32

nburnett, 2020-08-25:
For sure fixed my problem! Thank You