armadsen/ORSSerialPort #102: Simple example using Arduino Uno + CommandLineDemo

marciok, 2016-03-01:
Hi guys, I'm trying to send a signal to my Arduino Uno, using the CommandLineDemo. However I don't get any response, for example :

  `>  0xF9`
 (It should report the protocol version)
I can see the board blinking, so It's connected, but I can't detect any response from the board.

Am I doing something wrong? excuse my ignorance, I'm just getting started with serial programming. 
Thanks!

armadsen, 2016-03-02:
Hi, it seems like you're seeing the same issue as Issue #99. See there for a detailed answer, but in short, it may be that you need to set the RTS line high:

```swift
port.RTS = true 
```

marciok, 2016-03-02:
Hi @armadsen thanks, for the reply. I also noticed that I was doing some things wrong, like not sending the complete byte array. 

Question: If I want to send a message --Ask for the firmware version, for example--  to my Arduino. That means I'm expecting an answer. Should I get the response by just sending the data thought the terminal? or the ideal way is to build and send a request?
Ultimately I'm trying to query my board using the Firmata protocol thought out the serial port.

I'll try again with the `port.RTS = true` option.  

Thanks.

armadsen, 2016-03-02:
If you're expecting a specific answer to your message, you should probably use the ORSSerialPort [Request/Response API](https://github.com/armadsen/ORSSerialPort/wiki/Request-Response-API). There's no reason you can't use that API in a command-line app, if that's what you'd like to do.

As far as Firmata goes, that's not something I have direct experience with, but I think you should be able to implement it using ORSSerialPort just fine. There's an Objective-C Firmata project here: https://github.com/fibasile/Firmata-ObjC . It uses ORSSerialPort, however it hasn't been updated in a couple years, and is using an old version of ORSSerialPort, from before the packet parsing and request/response APIs were added. Still, might be worth looking at.

marciok, 2016-03-07:
@armadsen Thanks, for the tip. By looking at Firmata-ObjC I was able to send/received data from my Arduino Uno, using ORSSerialPort.

armadsen, 2016-03-14:
Glad to hear it. Feel free to ask if you run into anything else.

closed 2016-03-14 by armadsen