armadsen/ORSSerialPort #103: ORSSerialRequest send retry

lm2s, 2016-04-28:
Hello,
Is there a simple way to add retry capabilities to ORSSerialRequest?
It might be good to have some kind of retransmission system, if a send fails, it's re-send X times until it actually fails.

armadsen, 2016-04-28:
If you're not queueing requests up using `ORSSerialPort`'s request queue (ie. you only send one request at a time, waiting for it to succeed before sending the next), it would be easy enough to implement this yourself. You could (ab)use the `userInfo` property to store a number of retries, then keep a count, and just resent failed requests in `-serialPort:requestDidTimeout:`.

However, if you're using the queue to fire off multiple requests at a time, it's a little more complicated, since a simple `-sendRequest:` call will put the request retry at the end of the queue. In that case, I suppose you could do something like:

```objective-c
-serialPort:(ORSSerialPort *)port requestDidTimeout:(ORSSerialRequest *)request
{
    if (shouldRetryRequest) { // logic to determine this omitted for simplicity
       NSArray *queue = port.queuedRequests;
       [port cancelAllQueuedRequests];
       [port sendRequest:request]; // Resend the failed request
       for (ORSSerialRequest *request in queue) { [port sendRequest:request]; } // Put queued requests back in the queue
   } else {
       // handle "real" failure
   }
}
```
(Caveats: I haven't tried this, and this code was typed off the top of my head in Safari.)

I'd be open to adding real API for this if it seems like a common enough thing to want to accomplish. A `numberOfTimesToRetry` property on `ORSSerialRequest` (defaulting to zero) might work well enough. In my (limited) experience, it's more common that failures are systemic, e.g. a cable is broken, the device on the other end crashed, etc., where a simple retry scheme isn't likely to be valuable. I'm interested in knowing what your set up looks like.

lm2s, 2016-04-29:
I'm building a program to update the firmware in a device. The update must be performed by splitting the firmware binary into chunks to be sent as packets, so if one fails without the retry mechanism all will fail.
It's true that I haven't had any issues where a retry mechanism would be useful (e.g., no packets actually failed). My suggestion comes more from a "playing it safe" perspective.
Since I still don't have the program in production being actually used I can't say that the retry mechanism is of some actual use and I do agree that it might not really provide any actual benefit.
I'll go ahead without the retry mechanism and will implement the algorithm you suggested if I find it necessary at some point.
You can close this. I'll keep you informed if I find a situation where the retry mechanism is of some benefit. 
Thank you for your time and attention.

armadsen, 2016-04-29:
Please do let me know if you end up thinking a real retry mechanism makes sense, and good luck with the program you're writing!

closed 2016-04-29 by armadsen