armadsen/ORSSerialPort #105: Xon/Xoff software flow control

jrl290, 2016-08-07:
Is there currently or any plans for Xon/Xoff flow control? Or is there an easy means to implement over ORSSerialPort?

Thank you,
James

armadsen, 2016-08-08:
Hi James,

No, I don't currently have any plans to add Xon/Xoff flow control. To be honest, this is the first time since releasing ORSSerialPort that anyone has asked for it. I've never needed to use it myself, either, and therefore don't fully understand all the details of it. However, I _believe_ it's purely implemented in software, which means you could implement it over ORSSerialPort by writing code that sent and watched for the Xon/Xoff bytes as appropriate.

I'm certainly not opposed to adding built in support for it, but it's not something I'm likely to do myself. If you're interested in implementing it, I'd love to review a pull request that implements it.

What are you doing that requires Xon/Xoff support?

jrl290, 2016-08-10:
I'm working with a serial to IR adapter. It uses the RTS line for power. Originally I thought that it also used the DTR line for power as well and software flow control. Turns out it uses DTR for flow control. So for me Xon/Xoff support is no longer necessary.
Thank you!

armadsen, 2016-08-10:
Nice, that makes life easier. I'm going to close this, then. Feel free to open new issues if you run into further questions.

closed 2016-08-10 by armadsen