armadsen/ORSSerialPort #106: Address Sanitizer reports use of deallocated memory in sendData:

kirk-kerekes, 2016-08-12:
Address Sanitizer _intermittently_ reports use of deallocated memory in **sendData**:, as invoked by **sendNextRequest**->**reallySendRequest:**. 
My guess is over-aggressive ARC activity -- the  request is deleted from the requests queue, and _then_ sent. 

Ought to be OK, but there are enough async activities floating around to make me wonder.

armadsen, 2016-08-12:
Hi Kirk,

Thanks very much for the report. I'll take a look at this. Either way, I'd rather not have address sanitizer complaining about anything!

armadsen, 2016-12-12:
So far I'm unable to reproduce this in the (Objective-C) RequestResponseDemo example project included with ORSSerialPort. Running the app for a while with Address Sanitizer enabled does not result in any flagged errors. @kirk-kerekes: Are you able to provide any further information about reproducing this?

kirk-kerekes, 2016-12-12:
Only that my app is likely more complicated and busier than the example app.

armadsen, 2018-07-02:
Closing this due to inability to reproduce it.

closed 2018-07-02 by armadsen