armadsen/ORSSerialPort #107: Using ORSSerialPort as Plugin for Unity3D (error receive mex)

alex920a, 2016-08-30:
While receiving messages I'm encountering this error:

2016-08-30 16:56:18.862 BuildOSX[772:64771] -[ORSSerialPortHelper serialPort:didEncounterError:] Bluetooth-Incoming-Port Error Domain=NSPOSIXErrorDomain Code=35 "Resource temporarily unavailable" UserInfo={NSFilePath=/dev/cu.Bluetooth-Incoming-Port, NSLocalizedDescription=Resource temporarily unavailable}

Any Idea?

armadsen, 2016-08-30:
I haven't actually used ORSSerialPort with Bluetooth ports, myself (though others have). I'm not exactly sure what my cause that error. Does it happen when you try to open the port, or mid-stream after you've opened it and started sending/receiving data?

alex920a, 2016-09-06:
ok I've resolved the problem. It was just a conflict with bluetooth peripherals.

closed 2016-09-06 by alex920a