armadsen/ORSSerialPort #110: Request: Support for Swift 3.0

sunapi386, 2016-09-23:
Issue for the folks trying to migrate to Swift 3.0: 

I see the milestone already exists, but is there a timeline for completing this? 
https://github.com/armadsen/ORSSerialPort/milestone/12

armadsen, 2016-09-24:
No, I have no real timeline for this. It's not necessary to use ORSSerialPort from Swift, as the Objective-C version bridges into Swift fine. Are you running into specific trouble using ORSSerialPort from Swift 3 code?

I have opened #111 for converting the existing Swift examples to Swift 3.0.

armadsen, 2016-09-24:
Just opened PR #112 to convert the Swift examples to Swift 3. I want to double check with fresh eyes before merging.

armadsen, 2016-12-12:
To recap: ORSSerialPort already works fine from Swift 3.0 as it's written in Objective-C. I have merged #112 which converts the included Swift examples to Swift 3.0. Eventually, ORSSerialPort will be rewritten in Swift, but until that that time new versions of Swift should not require any changes to ORSSerialPort itself.

closed 2016-12-12 by armadsen