armadsen/ORSSerialPort #114: On Sierra ORSSerialPortManager.shared().availablePorts shows bluetooth port

moog-man, 2016-10-29:
I've tried to use ORSSerialPortManager.shared().availablePorts, but the array has only one element which contains "Bluetooth-Incoming-Port" even though I have connected a game controller to one of the USB ports. The game controller shows up ok, when I run a system diagnose (About this Mac)

Controller:

  Produkt-ID:	0x028e
  Hersteller-ID:	0x045e  (Microsoft Corporation)
  Version:	1.10
  Geschwindigkeit:	Bis zu 12 MBit/s
  Hersteller:	SHANWAN
  Standort-ID:	0x14100000 / 9
  Verfügbare Stromstärke (mA):	500
  Zusätzlicher Betriebsstrom (mA):	0

When I run your demo code ORSSerialPortDemo (Objective C)  it also only shows the bluetooth device.

armadsen, 2016-10-29:
Are you sure the game controller is supposed to present itself as a serial port? What you've posted from System Profiler is information about it as a USB device. All connected USB devices show up in System Profiler, but that doesn't mean they're serial (e.g. RS-232) devices.

Am I correct in assuming that you're using an Xbox controller? If so, I think you have to talk to it directly over USB using IOKit or potentially something like [libusb](http://www.libusb.org). I did find [this project](https://github.com/amandravin/GamePad) which might be helpful. I have no experience in this area, myself.

moog-man, 2016-10-30:
Hi,

thanks for the reply. I have to admit, that I am also new to the subject of USB on Mac. It seems the controller I have is not working on Mac at all because neither the game, nor some joystick utilities recognize it. I've ordered a Logitech controller which is supposed to work on Mac.

Will keep you posted!

Mat

Matthias Schorer
Stiftsbogen 31
81375 München
t: +49 89 5796 7923
m: +49 170 542 4819
e: mschorer@me.com

sent from my iPhone 5

armadsen, 2016-12-12:
Closing due to inactivity.

closed 2016-12-12 by armadsen