armadsen/ORSSerialPort #119: Please consider supporting Carthage

JetForMe, 2017-06-25:
Please support [Carthage](https://github.com/Carthage/Carthage) dependency management. It's cleaner to use than Cocoapods. Thank you.

armadsen, 2017-06-25:
Hi @JetForMe,

ORSSerialPort already supports Carthage as noted in the [How To Use ORSSerialPort](https://github.com/armadsen/ORSSerialPort#how-to-use-orsserialport) section of the README, as well as the [Installation Guide](https://github.com/armadsen/ORSSerialPort/wiki/Installing-ORSSerialPort#carthage).

If you have trouble getting Carthage working for ORSSerialPort, please open a new issue.

closed 2017-06-25 by armadsen

JetForMe, 2017-06-25:
I'm so sorry! I should've look closer. I skimmed the docs and didn't see it, and admittedly looked for the Carthage badge and didn't see that, either. Really sorry for the noise!