armadsen/ORSSerialPort #121: Can not get the product ID and vendor ID

liyueMaster, 2017-09-19:
https://github.com/armadsen/ORSSerialPort/wiki/Getting-Vendor-ID-and-Product-ID
The method in this page can not get the product ID and vendor ID.  IORegistryEntryCreateIterator got an error result and return nil for property.

martinjbaker, 2017-09-27:
I'm seeing the same thing when I disconnect the USB device and connect it again. Works fine for the first connection.

netizen01, 2018-10-29:
Anyone else been able to fix this?

underscoreforbidden, 2019-03-08:
Based on https://stackoverflow.com/questions/34464672/make-a-list-of-serial-usb-device-with-specific-ids

``` objectiveC 
- (int)VendorID
{
    // Variable declaration
    int vid;
    CFTypeRef cf_vendor;
    
    vid = -1;
    // Search properties among parents of the current port
    cf_vendor = IORegistryEntrySearchCFProperty(_IOKitDevice, kIOServicePlane,
                                                CFSTR(kUSBVendorID),
                                                kCFAllocatorDefault,
                                                kIORegistryIterateRecursively | kIORegistryIterateParents);
    
    // Decode & print VID
    if (cf_vendor && CFNumberGetValue(cf_vendor , kCFNumberIntType, &vid))
    {
        printf("\nUSB VID: %04X", vid);
    }
    
    // Release CFTypeRef
    if (cf_vendor)  CFRelease(cf_vendor);
    
    // Return vendor id
    return vid;
}
```  

``` objectiveC 
- (int)ProductID
{
    // Variable declaration
    int pid;
    CFTypeRef cf_product;
    
    pid = -1;
    // Search properties among parents of the current port
    cf_product = IORegistryEntrySearchCFProperty(_IOKitDevice, kIOServicePlane,
                                                CFSTR(kUSBProductID),
                                                kCFAllocatorDefault,
                                                kIORegistryIterateRecursively | kIORegistryIterateParents);
    
    // Decode & print PID
    if (cf_product && CFNumberGetValue(cf_product , kCFNumberIntType, &pid))
    {
        printf("\nUSB PID: %04X", pid);
    }
    
    // Release CFTypeRef
    if (cf_product)  CFRelease(cf_product);
    
    // Return product id
    return pid;
}
```

dekosan, 2019-03-13:
Works fine for me as described by underscore forbidden but I experienced the same problem as martinjbakker: disconnecting / reconnecting the USB device gives a (null) PID/VID. So I included a workaround: restart the app when removing an USB device. I would be glad to see a solution for the cause of the problem which I couldn't find.

- (void)relaunchAfterDelay:(float)seconds
{
    NSTask *task = [[NSTask alloc] init];
    NSMutableArray *args = [NSMutableArray array];
    [args addObject:@"-c"];
    [args addObject:[NSString stringWithFormat:@"sleep %f; open \"%@\"", seconds, [[NSBundle mainBundle] bundlePath]]];
    [task setLaunchPath:@"/bin/sh"];
    [task setArguments:args];
    [task launch];
    [[NSApplication sharedApplication] terminate:nil];
}
- (void)serialPortsWereDisconnected:(NSNotification *)notification
{
	NSArray *disconnectedPorts = [notification userInfo][ORSDisconnectedSerialPortsKey];
	NSLog(@"Ports were disconnected: %@", disconnectedPorts);
	[self postUserNotificationForDisconnectedPorts:disconnectedPorts];
        [self relaunchAfterDelay:0.1];
}

armadsen, 2019-04-08:
Unfortunately, I don't know why getting these fails after a disconnect/reconnect. I don't use the PID/VID for anything myself, so haven't noticed this. I'll look into it at some point, though.

KevinVitale, 2019-04-25:
Should be fixed by #144.