armadsen/ORSSerialPort #123: Not working with Sparkfun Pro Micro

pierdr, 2017-11-02:
Hi,
with a [Sparkfun Pro Micro](https://www.sparkfun.com/products/12640) that ships with a Atmel MEGA 32u4 I'm not able to receive any incoming data. 

The code (arduino) I'm testing is the following:
```
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println(millis());
  delay(2000);
}
```

It works with an Arduino seamlessly. With the Pro Micro this [`line`](https://github.com/armadsen/ORSSerialPort/blob/e1aca2bf5c2a92039e234f68749ab6a774bd5625/Source/ORSSerialPort.m#L294) is never reached. 

Actually there is a case in which the line is reached and it happens when a second serial application (cool term in my case) opens the serial communication with the pro micro; when open with an other program the data is received by both my application and cool term.

It seems to me that the connection is never really open (even if the open delegate method is triggered) and the data is not received. 
Any ideas on what the problem might be?

pierdr, 2017-11-07:
UPDATE: 
I managed to run the Sparkfun Pro Micro (ATMEGA 32u4) but using the low level API from the Apple documentation ([here](https://developer.apple.com/library/content/samplecode/SerialPortSample/Listings/SerialPortSample_SerialPortSample_c.html#//apple_ref/doc/uid/DTS10000454-SerialPortSample_SerialPortSample_c-DontLinkElementID_4))
It will be great to make this happen in ORSSerialPort!

>  _serialFileDescriptor = open(
>                                  [address UTF8String],
>                                  O_RDWR |
>                                  O_NOCTTY |
>                                  O_NONBLOCK );
>     // block non-root users from using this port
>     ioctl(_serialFileDescriptor, TIOCEXCL);
>     
>     // clear the O_NONBLOCK flag, so that read() will
>     //   block and wait for data.
>     fcntl(_serialFileDescriptor, F_SETFL, 0);
>     
>     // grab the options for the serial port
>     tcgetattr(_serialFileDescriptor, &options);
>     
>     // setting raw-mode allows the use of tcsetattr() and ioctl()
>     cfmakeraw(&options);
>     
>     // specify any arbitrary baud rate
>     ioctl(_serialFileDescriptor, IOSSIOSPEED, &baudRate);
>     
>     // start a thread to read the data coming in
>     [self
>      performSelectorInBackground:
>      @selector(incomingTextUpdateThread:)
>      withObject:
>      [NSThread currentThread]];

justinisdumb, 2018-03-06:
I am having the exact same problem with the Adafruit Itsy bitsy.  

It seems to be able to send data over the serial connection, but it never receives any responses.  
The delegate never gets triggered.  It works perfectly with an Arduino Uno board.

armadsen, 2018-03-06:
I've just ordered an Itsy Bitsy so I can look into this. It probably won't be here until next week sometime, so if you come up with something in the meantime, please let me know.

justinisdumb, 2018-03-08:
Thanks for such a quick response!  
Im not sure where to dig into it to look for the issue, but let me know if there is anything you need from me.  I appreciate your help, this framework is insanely helpful.

justinisdumb, 2018-03-27:
Hey armadsen,  I just wanted to check in and see if you got a chance to look at the Itsy Bitsy issue.
Thanks

justinisdumb, 2018-07-18:
Hey armadsen, just checking in.  Any way I can help in debugging?  Id love to switch all my systems over to the Itsy Bitsy.  Thanks!

justinisdumb, 2018-10-02:
Not sure if you ever got a chance to look into this, but I have found that when I start the demo command line app, I can't get a connection or response from the Itsy Bitsy.  If I then open up the "Serial Monitor" from the Arduino app, the connection comes to life and the demo app starts seeing the board.  Not sure if that might help.

Dirk-, 2022-03-04:
@pierdr 
> UPDATE: I managed to run the Sparkfun Pro Micro (ATMEGA 32u4) but using the low level IOKit/ioctl API from the Apple documentation ([here](https://developer.apple.com/library/content/samplecode/SerialPortSample/Listings/SerialPortSample_SerialPortSample_c.html#//apple_ref/doc/uid/DTS10000454-SerialPortSample_SerialPortSample_c-DontLinkElementID_4)) It will be great to make this happen in ORSSerialPort!
> 
> > _serialFileDescriptor = open(
> > [address UTF8String],
> > O_RDWR |
> > O_NOCTTY |
> > O_NONBLOCK );
> > // block non-root users from using this port
> > ioctl(_serialFileDescriptor, TIOCEXCL);
> > ```
> > // clear the O_NONBLOCK flag, so that read() will
> > //   block and wait for data.
> > fcntl(_serialFileDescriptor, F_SETFL, 0);
> > 
> > // grab the options for the serial port
> > tcgetattr(_serialFileDescriptor, &options);
> > 
> > // setting raw-mode allows the use of tcsetattr() and ioctl()
> > cfmakeraw(&options);
> > 
> > // specify any arbitrary baud rate
> > ioctl(_serialFileDescriptor, IOSSIOSPEED, &baudRate);
> > 
> > // start a thread to read the data coming in
> > [self
> >  performSelectorInBackground:
> >  @selector(incomingTextUpdateThread:)
> >  withObject:
> >  [NSThread currentThread]];
> > ```

The only real differences I see are `ioctl(_serialFileDescriptor, TIOCEXCL);` and that you didn't set `O_EXLOCK` on the `open()` call.
I changed `ORSSerialPort.m` to reflect that, but it did not help in my case.

@armadsen It sees the library lost its ability to communicate with Arduino boards as well in recent macOS versions. The fact that it works after the port is opened in another application makes me think there is something missing in the serial port or GCD initialization in ORSSerialPort. I looked into the source but did not find any clues.

armadsen, 2022-03-04:
Thanks for the followup report, @Dirk-. I've left myself a note to try digging into this this weekend.

Dirk-, 2022-03-17:
I found a solution:

Somehow, `RTS` and `DTR` have to be set to `TRUE` for a successful connection with an Arduino on recent macOS versions.

In the `openOrClosePort` method of ORSSerialPortDemo, for example, you could do this:

```
- (IBAction)openOrClosePort:(id)sender
{
    self.serialPort.isOpen ? [self.serialPort close] : [self.serialPort open];
    // For Arduino:
    if (self.serialPort.isOpen) {
        self.serialPort.RTS = TRUE;
        self.serialPort.DTR = TRUE;
    }
}

```
It is not sufficient to set `usesDTRDSRFlowControl` or `usesRTSCTSFlowControl` to `TRUE`. No need to change these flags.