armadsen/ORSSerialPort #130: Doesn't communicate with Adafruit ItstyBitsy 32u4

jwalk487, 2018-10-23:
Hi,

I am having trouble communicating with Adafruit ItstyBitsy 32u4. When I open the port it shows that the port has been opened successfully but I cannot send or receive data.

If I use Serial Monitor in the Arduino application I can communicate with no issue. I also have no problem communicating using a python script. 

If I open Arduino Serial Monitor AND connect with ORS it seems to work fine and I can send and receive.

It seems ORS is just not opening the opening the connection properly.

I have also tried ORS with an Arduino Uno and that works fine so I am not sure what's causing this issue.

Thanks for your help

Dirk-, 2022-03-04:
Same behavior with Arduino Nano 33 BLE and ASURO robots. It seems to have gotten worse in recent macOS system versions.

pierdr, 2022-09-27:
I have the same issue with a 32u4. 
The incoming data starts if I open the serial port with another program (in my case CoolTerm, if I try to open the stream with Arduino doesn't work). 

I believe there is something on the low level that is done incorrectly (coolterm is triggering the data to start come in).

pierdr, 2022-09-27:
Sorry this problem is actually solved already. See thread: https://github.com/armadsen/ORSSerialPort/issues/123

Setting RTS and DTR to true makes the stream of incoming serial data start:
```
func serialPortWasOpened(_ serialPort: ORSSerialPort) {
        print("port was opened");
        serialPort.rts = true;
        serialPort.dtr = true;
        
    }
```