armadsen/ORSSerialPort #134: Serial Break

jdelaune, 2019-02-04:
Is it possible to send a serial break?

Thanks

armadsen, 2019-02-04:
Not directly with the `ORSSerialPort` API, no. To be honest, this is the first time this question has come up. You may be able to use `tcsendbreak()` on the underlying file descriptor for the serial port, but that's not something I've tested myself. The file descriptor is available in the `fileDescriptor` property on `ORSSerialPort`. That's a "private" property, not exposed in the public interface, but you could redeclare it in a category to access it, at least to try this approach out. If you get it working, I'd love to know about it.

armadsen, 2019-02-07:
Closing as answered.

closed 2019-02-07 by armadsen