armadsen/ORSSerialPort #150: Debug transmission

jdelaune, 2019-11-22:
Hi thanks for the great library.

I'm having issues with a particular serial device. Using CoolTerm or SwiftSerial I can send commands to it fine. However using ORSSerialPort it never works. I checked all the port settings and I've checked the raw data I pass to `send()` and its exactly what is sent using CoolTerm and SwiftSerial. So I'm wondering if something else is being sent as well in the internals, and how I can debug this? ORSSerialPort works fine with all other serial devices I've used it with in the past. It's just this one device which is a bit sensitive it seems.

Any pointers appreciated.

armadsen, 2019-11-22:
Can you tell me details of the device? Also, does it just silently fail to send anything? Or does it send garbage? Something else?

jdelaune, 2019-11-22:
It's a Tenma 72-2540 bench power supply. Just trying a simple example of sending `OUT1` ascii encoded. Which should turn it on. `send()` returns `true`. Unfortunately the Tenma doesn't have anything for me to see what is happening which is why I wondered if I can debug what is being sent.

It might not be in the send command, it could be something which is sent or setup during `open()` I guess. I just know the device is very sensitive, send it something it's not expecting it will lock up :)

Thanks

closed 2022-05-07 by jdelaune