armadsen/ORSSerialPort #152: 'IOKit/IOTypes.h' file not found

sree-86, 2020-01-06:
Hey, thanks for the library. 

I am having this issue while using this library via Swift Package Manager.

When I compile my project, I get the following error:

`2. Did not find header 'IOTypes.h' in framework 'IOKit' (loaded from '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.2.sdk/System/Library/Frameworks')`

The error occurs at: 

> ORSSerialPort.h:28

sree-86, 2020-01-06:
I also get another error:

`'IOKit/serial/IOSerialKeys.h' file not found`

at

> ORSSerialPort.m:30

armadsen, 2020-01-06:
Hi there,

Unfortunately, ORSSerialPort is a macOS library and does not work on iOS, where the required APIs for serial port access are not public (and are different).

sree-86, 2020-01-06:
I know that. My target device is "My Mac".

armadsen, 2020-01-06:
Ahh, OK. The build system is looking in the iOS SDK for some reason. I don't know of a reason that that would happen unless your project is configured to build for iOS. Are you building with swiftpm (on the command-line) or are you using Xcode?

Either way, I'm happy to take a look if you're able to share the project with me (andrew@openreelsoftware.com).

sree-86, 2020-01-06:
I am using your Swift Package as a dependency for my Swift Package which has iOS and Mac OS platforms. I built it using the command line and then used Xcode to build my package and that's when I got the error.

Earlier when I used to import your framework, it was working fine since I was using different targets for iOS and MacOS in my library. But, I recently updated my library to a Swift Package and I'm unable to assign different platforms to different Package targets.

sree-86, 2020-01-06:
I removed your dependency, then I performed a clean build and added the dependency again and it works now. Maybe it's an Xcode bug since my Xcode crashed several times while trying to remove your dependency. Thanks a lot anyway. Have a great day!

closed 2020-01-06 by sree-86

kjyv, 2020-08-13:
@armadsen Hey, sorry to revive this closed ticket but you mention the iOS non-public API for reading serial port values. I can't find anything anywhere about it. Do you have more info on this? It would be very useful for a project we're working on. Thanks!

armadsen, 2020-08-13:
Hi @kjyv,

In order to use the APIs I was referring to in an app on the app store, you need to be a member of [Apple's MFi](https://developer.apple.com/programs/mfi/) (Made for iPhone/iPad) program. If you're doing a one-off, you can buy Redpark's [serial adapters](https://redpark.com/products/serial-adapters/) and they provide an SDK. However, to my knowledge, you can only use it for local/development, not in an app store app unless you're a member of the MFi program.

kjyv, 2020-08-14:
@armadsen Thanks, I knew about the MFi program but I was asking about a possible non-public API to use the USB HID Serial profile. I don't want to use it in an app on the app store so it could be unofficial and should work without MFi certified devices. (This is not for apps with jailbreaks but for a prototype/internal use device). Redpark seems to only be available for Lightning, not USB-C.

armadsen, 2020-08-14:
When I said non-public, I was indeed referring to the MFi-gated APIs. I don't know of a solution for USB-C iPads, unfortunately. If I had to make something work there, I'd probably use a Bluetooth <-> Serial adapter of some sort. There are several out there. I'd probably start with the Airconsole: http://www.get-console.com/airconsole-le/

kjyv, 2020-08-15:
Ok, I see. MFi should also work with USB-C as I understand but the certification process is probably a lot of work and might not even be approved for non-public hardware. And our connection needs to be low latency (audio), so everything should be wired. We're looking into coding the data into HID joystick commands and see how big the bandwidth is.