armadsen/ORSSerialPort #156: Operation not permitted

tangzhifengjluzh, 2020-03-27:
Uploading an application requires opening the sandbox, but when opening sandbox permissions, an error is reported

Serial port usbserial-14630 encountered an error: Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted" UserInfo={NSFilePath=/dev/cu.usbserial-14630, NSLocalizedDescription=Operation not permitted}

sccxx, 2020-03-28:
https://github.com/armadsen/ORSSerialPort/wiki/Installing-ORSSerialPort#use-in-sandboxed-applications

Try add "com.apple.security.device.serial" to the entitlements file.

armadsen, 2020-03-28:
As sccxx's answer says, you need to make sure you've included the `com.apple.security.device.serial` to your sandbox entitlements file. If this doesn't fix the issue, let me know, but I'm going to close this for now.

closed 2020-03-28 by armadsen

tangzhifengjluzh, 2020-03-30:
> https://github.com/armadsen/ORSSerialPort/wiki/Installing-ORSSerialPort#use-in-sandboxed-applications
> 
> Try add "com.apple.security.device.serial" to the entitlements file.

Thank you very much for solving this problem perfectly

tangzhifengjluzh, 2020-03-30:
Thank you very much for solving this problem perfectly