armadsen/ORSSerialPort #158: Having Issues with ORSSerialPort in Xcode 11 for Mac OS 10.15

gchapel1, 2020-05-30:
As sccxx's answer says, you need to make sure you've included the `com.apple.security.device.serial` to your sandbox entitlements file. If this doesn't fix the issue, let me know, but I'm going to close this for now.

_Originally posted by @armadsen in https://github.com/armadsen/ORSSerialPort/issues/156#issuecomment-605527428_

I am having issues opening serial ports using ORSSerialPort.  I am using the serialPortManager to get a list a ports, then allowing a user to select the port to open and the baud rate for the port.  Then the user presses the Open Button to open the port.  However every time I keep getting the following error.

"SerialPort usbmodem903NTKF843172 encountered an error: Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted" UserInfo={NSFilePath=/dev/cu.usbmodem903NTKF843172, NSLocalizedDescription=Operation not permitted}"

I noticed that you supplied the above answer to another person who has a similar issue.  However I am using Mac OS 10.15 and Xcode 11.  I have never used the entitlements section before and I didn't have this issue before when I wrote a similar app using objective-C.  I am using Swift for this app and I can't get it to work.

Any ideas as to why? The NSFilePath appears to be correct.  The baud rate has a proper setting.  The code compiles without errors.  But the serial port fails to open and the above error message is printed in the debug window.

Please help.  I am new to coding and I am a bit lost on this issue.

yxq-Jason, 2020-06-02:
> Originally posted by @armadsen <https://github.com/armadsen> in #156 (comment) <https://github.com/armadsen/ORSSerialPort/issues/156#issuecomment-605527428>

Hi, I am also use Xcode 11.3 and MacOS 10.15, but when I create new App with swift or Objective-C, the Application source will all generate entitlements file automatically(I don’t know what’s the difference setting in ur Xcode).

And same issue I met before and fixed:

Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted"

—> Check sandbox and disable, then port open works.

gchapel1, 2020-06-07:
Thanks.  Yes I managed to get it working. Xcode was being flakey I guess and wasn't seeing my entitlements file. I removed the sandbox completely and I was fine.  Then I added the sand box got out of the app and my entitlements file appeared.  I entered the property as instructed and all was well.

closed 2020-06-07 by armadsen