armadsen/ORSSerialPort #159: Can't get application to find ORSSerial libraries when archiving app for testing

gchapel1, 2020-06-07:
Please help.  I have developed an app using Xcode 11 that uses the ORSSerialPort libraries for accessing the serial port.  The application is built in Swift  and I installed the libraries exactly as described in the Installation guide and the online video. 

I can build without errors and run my application from within Xcode without issues.  However now I am trying to archive the build and distribute the application for testing.  I can successfully archive the app with Xcode 11 but every time I run the product it crashes with the following error.

"Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/ORSSerial.framework/Versions/A/ORSSerial
  Referenced from: /Users/USER/Desktop/*/Scanner Remote.app/Contents/MacOS/Scanner Remote
  Reason: image not found"

It seems to be complaining about missing library for ORSSerial.  

Any ideas how I can fix this?  I don't understand why the archive is not including the libraries in the project file. 

Please help.  I have finally figured out all the issues with my application and it works great from within Xcode.  Now all I have to do is get it out.

armadsen, 2020-06-07:
Sounds like you don't have your project configured to embed MIKMIDI.framework in the finished .app bundle. Make sure that in the "Frameworks, Libraries, and Embedded Content" section of the General tab of settings for your app target, you have "Embed & Sign" selected as the Embed setting for MIKMIDI.framework.

[image]

gchapel1, 2020-06-07:
Hi Andrew,

Thank you for the repose.  I just discovered this late last night.  

However, I think you are showing me the wrong libraries by accident. 

The library framework that I needed to embed was ORSSerial.framework as this is the framework I am using in my app.

But thank you for the response as I wasn’t sure about the embed without signing and embed with signing option.

If I get this app released on the App Store, what test do I need to include and where do I put it for acknowledging the ORSSerial libraries that you wrote.

My app will be a free app but I wanted to acknowledge your great set of library functions for serial port communication, somewhere  in the app.

Greg

armadsen, 2020-06-07:
Ahh, right! I maintain both MIKMIDI and ORSSerialPort and didn't pay close enough attention when I went to make the screenshot.

As for properly crediting the project, you can put it wherever it makes sense. In my projects, I have a "Copyright Info" item in the Help menu that shows a window with a text view containing license text and credits for the libraries I use.

For a lighter weight solution, just include a credit in the about box. Something like "Uses ORSSerialPort by Andrew Madsen", with a link to the license file (https://github.com/armadsen/ORSSerialPort/blob/master/LICENSE).

If you add a Credits.rtf file to your app’s resources, it will automatically get picked up by the default about box machinery and be used to populate text in that about window. This isn’t documented in any super clear way, though it is technically documented here: https://developer.apple.com/documentation/appkit/nsapplication/aboutpaneloptionkey/2869609-credits . It used to be in the default Mac app template in Xcode, but they took it out for some unknown reason. Attached is a little project that shows this in action (see screenshot too). It really is as simple as creating Credits.rtf (name and case is important here) in e.g. TextEdit, then adding it to your Xcode project.

[image]

[AboutDemo.zip](https://github.com/armadsen/ORSSerialPort/files/4743157/AboutDemo.zip)

gchapel1, 2020-06-07:
Ok, thank you Andrew. I will add the comment and the link in my “about” tab as you suggest. 

One more question if you can help me as I am a newbie.

Recently, I don’t know how it happened but I accidentally lost  the ORSSerial.xcodeproj subtree from my project.  It had something to do with when I embedded and then built again without embedding the frameworks.  I don’t know how it happened but it went blank.

I tried going to the console and typing  “git submodule add https://github.com/armadsen/ORSSerialPort.git Frameworks/ORSSerialPort” to get another copy but git responds with 

"A git directory for 'Frameworks/ORSSerialPort' is found locally with remote(s):
  origin	https://github.com/armadsen/ORSSerialPort.git
If you want to reuse this local git directory instead of cloning again from
  https://github.com/armadsen/ORSSerialPort.git
use the '--force' option. If the local git directory is not the correct repo
or you are unsure what this means choose another name with the '--name' option.

Do you know how I repair this issue and get your libraries back into my Xcode project?

Greg

gchapel1, 2020-12-04:
Hi Andrew,

I have been developing an app named "Scanner Remote” that is in the Mac App Store and it uses your ORSSerial libraries to open and read from the Mac Serial device. 

It works great and I have had no issues until I upgraded to Xcode 12.2 and tried to create a new project and link to your libraries using the exact same procedure that I always used in the past.  I use you instructions under the “Using an Xcode subproject” .   The code appears to build but fails when run with the following error message in my Xcode console:

dyld: Library not loaded: @rpath/ORSSerial.framework/Versions/A/ORSSerial
  Referenced from: /Users/greg/Library/Developer/Xcode/DerivedData/TestSerialBuild-anyhcydqtjnwgnfugzsykhoonkgm/Build/Products/Debug/TestSerialBuild.app/Contents/MacOS/TestSerialBuild
  Reason: no suitable image found.  Did find:
	/Users/greg/Library/Developer/Xcode/DerivedData/TestSerialBuild-anyhcydqtjnwgnfugzsykhoonkgm/Build/Products/Debug/ORSSerial.framework/Versions/A/ORSSerial: code signature in (/Users/greg/Library/Developer/Xcode/DerivedData/TestSerialBuild-anyhcydqtjnwgnfugzsykhoonkgm/Build/Products/Debug/ORSSerial.framework/Versions/A/ORSSerial) not valid for use in process using Library Validation: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)
	/Users/greg/Library/Developer/Xcode/DerivedData/TestSerialBuild-anyhcydqtjnwgnfugzsykhoonkgm/Build/Products/Debug/ORSSerial.framework/Versions/A/ORSSerial: stat() failed with errno=1

Here is the output I see in Xcode on where the error takes place:

dyld`__abort_with_payload:
    0x100071ed4 <+0>:  movl   $0x2000209, %eax          ; imm = 0x2000209 
    0x100071ed9 <+5>:  movq   %rcx, %r10
    0x100071edc <+8>:  syscall 
->  0x100071ede <+10>: jae    0x100071ee8               ; <+20>
    0x100071ee0 <+12>: movq   %rax, %rdi
    0x100071ee3 <+15>: jmp    0x100070408               ; cerror_nocancel
    0x100071ee8 <+20>: retq   
    0x100071ee9 <+21>: nop    
    0x100071eea <+22>: nop    
    0x100071eeb <+23>: nop 

Can you please take a look at this and see what might be the issue.  Since it sites some complaint about the ORSSerial not having a valid Team ID, etc, I thought it might have to do with some security issue new in 12.2 that I need to change somewhere?

Just so that you can see the issue more clearly, I have created a short screen shot video showing the procedure I used and the issue that ensues.  Here is the dropbox link for the video 

Video Link ===>. <https://www.dropbox.com/s/1xlhqx1bmme6zrf/Screen%20Recording%202020-12-03%20at%205.16.30%20PM.mov?dl=0 <https://www.dropbox.com/s/1xlhqx1bmme6zrf/Screen%20Recording%202020-12-03%20at%205.16.30%20PM.mov?dl=0>>

I really appreciate any info you could provide to assist with this issue.

Thanks,
Greg Chapel

gchapel1, 2020-12-04:
Hi Andrew,

I figured out how to fix the issue.  Apparenty, under the General tab of the Xcode build settings page, I needed to select “Embed & Sign” for the frameworks, libraries, and embedded content section.  This is a change because before I had always used “Do not Embed” and it compiled and ran fine.

This appears to be a change in 12.2.

Thanks,
Greg

armadsen, 2020-12-04:
Hi @gchapel1,

Glad you got it figured out. For what it's worth, there may have been a change that caused builds run from Xcode 12.2 to start failing when the framework isn't embedded, but it has always been required to embed them if you're going to distribute the app, so keeping that setting on Embed & Sign is ~always what you should do.

closed 2020-12-04 by armadsen