armadsen/ORSSerialPort #162: ORSSerial and SwiftUI/Combine

janhendry, 2020-10-01:
I pushed an entry on stackoverflow, where I had problems integrating ORSSerial into SwiftUI / Combine. I tried to get around the delegate pattern without success.  But I found a good way to integrate ORSSerialPort into a SwiftUI view. I would like to show them here. Maybe it will help someone, or maybe someone has an even better idea.

I have the a ViewStruct,  a ViewModel and then SerialPortCombine. The SerialPortCombine wraps the ORSSerialPort.

Start with a simple View
[image]

```
```
```
```
```
```
```
```
```
```
```
```
```
```
```
```

armadsen, 2020-10-01:
Thank you, this is great. I'd be happy to consider a pull request that adds your demo example to the examples folder in the repo.

janhendry, 2020-10-02:
pull request it out :)

ronnyandre, 2021-01-06:
I could really use this example app, so please add it in a pull request!