armadsen/ORSSerialPort #171: ORSSerialPort open  -1

949699582, 2021-01-21:
I use ORSSerialPort's open functtion,But the value of descriptor is -1. Here is the code!
descriptor = open([self.path cStringUsingEncoding:NSASCIIStringEncoding], O_RDWR | O_NOCTTY | O_NDELAY);

To check this error,I download the CoolTerm.And it display the port is connected!

Im confused!

armadsen, 2021-01-21:
Hi,

Have you verified that you've added the required sandbox entitlement for serial port access? If your app is sandboxed (the default Xcode Mac app template starts with sandboxing turned on), you'll need that or all attempts to open a serial port will fail. See here: https://github.com/armadsen/ORSSerialPort/wiki/Installing-ORSSerialPort#use-in-sandboxed-applications

949699582, 2021-01-21:
Thanks your reply:
I have already add "com.apple.security.device.serial" is "true" in entitlement. But the result of open function is still -1.

949699582, 2021-01-21:
Thanks you tips:
I delete sanbox,and open success

closed 2021-01-22 by armadsen