armadsen/ORSSerialPort #182: Add Recent or Time Sort Order to history list

eli104, 2021-08-31:
The only sorting I can find for the received history is by the number of Hits. Would it be possible to add a Sort Order to each column AND (more important), Sort by Recent or by Time received?

Reason: When I hear but do not see a call (and the display starts to scan), I cannot determine where the last message received was from. Would be great to see a list of calls in the order they were received (Recent Ascending/Descending).

armadsen, 2021-08-31:
Hi there,

I'm not exactly sure what you're asking about. It seems to be about a particular piece of ham radio software. This page is for ORSSerialPort, which is a serial port code library used by Mac software developers, rather than any particular end-user application.

I write Aether (a ham radio logging app), but I don't think you're asking about that.

-Andrew

eli104, 2021-09-02:
Sorry, it appears I may have jumped to the wrong branch.

closed 2021-09-02 by eli104