armadsen/ORSSerialPort #184: 'ORSSerial/ORSSerialPacketDescriptor.h' file not found

Janldeboer, 2022-03-08:
When using this package with the swift package manager, this error occurs.
It can easily fixed by removing "ORSSerial/" from the path (leaving "<ORSSerialPacketDescriptor.h>"), but this seems to break the package as used in the ORSSerialPortDemo Example.
How can this issue be resolved?

armadsen, 2022-03-08:
Hi Jan,

Thanks for the bug report. Can you try importing the current release version, 2.1.0? That works fine for me, and the change provoking this error happened after that release. I'll work on fixing this, but in the meantime, that may be a viable workaround.

Janldeboer, 2022-03-10:
Thanks, it works fine with 2.1.0 specified as version rule. I used the branch as a version rule before, that doesn't seem to be sufficient...

armadsen, 2022-12-17:
Fixed by #189

closed 2022-12-17 by armadsen