armadsen/ORSSerialPort #185: Port Connect/Disconnect Notifications break in Xcode 13

sethtc, 2022-04-07:
I believe, the new changes in notification permissions, and construction styles breaks the notification functionality of this framework.  I had the notifications working before update to Xcode 13, then after, they broke.  I am not smart enough to prescribe a fix, but have suspicions.  I think perhaps because the backend ORSSerial.m files are Objective-c and build notifications using Obj-C parlance they are not valid for Xcode 13 standards?  not sure. I will keep looking into fixes...

jaaguilar, 2022-04-24:
I'm seeing this as well. Is this project still active?

armadsen, 2022-04-24:
I’ve got this on my todo list to address. It’s not because the code is written in ObjC, however.
Pull requests are certainly welcome.

jaaguilar, 2022-04-24:
Glad to hear this project is still active! It's the best set of Serial APIs for macOS.
100% agree objc has nothing to do with this issue. 
I spent a few minutes debugging this issue today. Will keep digging. I'll let you know if I do find anything. :)

jaaguilar, 2022-04-25:
Notifications are working fine when I include ORSSerialPort in a brand new test project. 
It's only not working in a SwiftPackage I created, so I'm guessing it's just some entitlement issue.