armadsen/ORSSerialPort #186: Hex Received data is not continuous

sonwa, 2022-04-24:
Hi,
Sometimes the received data is not continuous.

[image]

sonwa, 2022-04-24:
Serial device data length is 5, and sometimes data is received separately. Not the complete received data.

armadsen, 2022-04-24:
Hi @sonwa. This is expected behavior. The underlying serial port hardware, the driver, and ORSSerialPort itself have no way of knowing what "complete" means for your data. Serial data isn't intrinsically structured, it's just a stream of bytes delivered as the port receives them.

In order to help you solve this, ORSSerialPort includes a packet parsing API. You can find information about how to use it on this Wiki page: https://github.com/armadsen/ORSSerialPort/wiki/Packet-Parsing-API. Let me know if you have questions as you start using it.

closed 2022-04-24 by armadsen